cartographer-project/cartographer_turtlebot #21: About the localisation mode

JiayanQ, 2016-10-06:
Thanks for your brilliant work!

I just wonder that can I use the well-built map by the SLAM mode to carry out a navigation or localisation work? Because I did not found the code to change the mode for SLAM to Localisation in the cartographer_node_main.cc. Should I modify the code in some other documents to achieve the navigation task? Thanks.

And thanks again for this amazing work!!

damonkohler, 2016-10-06:
Pure localization is not yet supported. However, lifelong mapping, which would negate the usefulness of pure localization, is planned.

The /map topic publishes a nav_msgs/OccupancyGrid that can be used by existing ROS nodes (e.g. navigation and localization).

closed 2016-10-06 by damonkohler

JiayanQ, 2016-10-06:
Thanks a lot for your reply.

Cheers,
Jiayan