cartographer-project/cartographer_turtlebot #23: An error occurred while I was running turtlebot with rplidar

oya123, 2016-10-09:
I want to use cartographer with turtlebot and rplidar, but I get the following error:
started core service [/rosout]
process[rplidarNode-2]: started with pid [14772]
process[robot_state_publisher-3]: started with pid [14776]
process[diagnostic_aggregator-4]: started with pid [14777]
process[mobile_base_nodelet_manager-5]: started with pid [14783]
process[mobile_base-6]: started with pid [14788]
process[cmd_vel_mux-7]: started with pid [14791]
process[turtlebot_laptop_battery-8]: started with pid [14804]
process[capability_server-9]: started with pid [14812]
process[app_manager-10]: started with pid [14826]
process[master-11]: started with pid [14831]
process[interactions-12]: started with pid [14836]
process[zeroconf/zeroconf-13]: started with pid [14840]
process[cartographer_node-14]: started with pid [14843]
process[flat_world_imu_node-15]: started with pid [14847]
[ INFO] [946692925.919112963]: I0101 10:15:25.000000 14843 configuration_file_resolver.cc:40] Found '/home/ubuntu/google_cartographer/install_isolated/share/cartographer_turtlebot/configuration_files/turtlebot_rplidar_2d.lua' for 'turtlebot_rplidar_2d.lua'.
[ INFO] [946692925.920156380]: I0101 10:15:25.000000 14843 configuration_file_resolver.cc:40] Found '/home/ubuntu/google_cartographer/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [946692925.920353547]: I0101 10:15:25.000000 14843 configuration_file_resolver.cc:40] Found '/home/ubuntu/google_cartographer/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [946692925.920969130]: I0101 10:15:25.000000 14843 configuration_file_resolver.cc:40] Found '/home/ubuntu/google_cartographer/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [946692925.921240130]: I0101 10:15:25.000000 14843 configuration_file_resolver.cc:40] Found '/home/ubuntu/google_cartographer/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [946692925.921785297]: I0101 10:15:25.000000 14843 configuration_file_resolver.cc:40] Found '/home/ubuntu/google_cartographer/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [946692925.922093213]: I0101 10:15:25.000000 14843 configuration_file_resolver.cc:40] Found '/home/ubuntu/google_cartographer/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [946692925.922635880]: I0101 10:15:25.000000 14843 configuration_file_resolver.cc:40] Found '/home/ubuntu/google_cartographer/install_isolated/share/cartographer/configuration_files/sparse_pose_graph.lua' for 'sparse_pose_graph.lua'.
[ INFO] [946692925.922895547]: I0101 10:15:25.000000 14843 configuration_file_resolver.cc:40] Found '/home/ubuntu/google_cartographer/install_isolated/share/cartographer/configuration_files/sparse_pose_graph.lua' for 'sparse_pose_graph.lua'.
[WARN] [WallTime: 946692926.055610] Battery : unable to check laptop battery info [/sys/class/power_supply/BAT0/charge_full_design || /sys/class/power_supply/BAT0/energy_full_design does not exist]
[ INFO] [946692926.378924547]: I0101 10:15:26.000000 14843 submaps.cc:176] Added submap 1
[ INFO] [946692926.991353963]: Zeroconf: service successfully established [turtlebot][_ros-master._tcp][11311]
[turtlebot_laptop_battery-8] process has finished cleanly
log file: /home/ubuntu/.ros/log/4cda3e64-bff1-11d3-9b70-7c5cf893e3bf/turtlebot_laptop_battery-8*.log
RPLidar health status : 0
[ INFO] [946692929.328739463]: I0101 10:15:29.000000 14843 cartographer_node_main.cc:615] gyro_link rate: unknown
[ INFO] [946692929.451284963]: I0101 10:15:29.000000 14843 motion_filter.cc:42] Motion filter reduced the number of scans to nan%.
[ INFO] [946692944.445147626]: I0101 10:15:44.000000 14843 cartographer_node_main.cc:615] base_footprint rate: 49.86 Hz 2.01e-02 s +/- 1.08e-03 s (pulsed at 114.16% real time)
[ INFO] [946692944.446515459]: I0101 10:15:44.000000 14843 cartographer_node_main.cc:615] camera_depth_frame rate: 6.24 Hz 1.60e-01 s +/- 9.11e-02 s (pulsed at 106.47% real time)
[ INFO] [946692944.447600209]: I0101 10:15:44.000000 14843 cartographer_node_main.cc:615] gyro_link rate: 106.72 Hz 9.37e-03 s +/- 2.31e-03 s (pulsed at 106.32% real time)
[ INFO] [946692959.496639372]: I0101 10:15:59.000000 14843 cartographer_node_main.cc:615] base_footprint rate: 49.87 Hz 2.01e-02 s +/- 1.94e-03 s (pulsed at 100.69% real time)
[ INFO] [946692959.498248372]: I0101 10:15:59.000000 14843 cartographer_node_main.cc:615] camera_depth_frame rate: 6.58 Hz 1.52e-01 s +/- 2.39e-02 s (pulsed at 99.88% real time)
[ INFO] [946692959.499091205]: I0101 10:15:59.000000 14843 cartographer_node_main.cc:615] gyro_link rate: 105.96 Hz 9.44e-03 s +/- 2.54e-03 s (pulsed at 100.17% real time)
F0101 10:16:10.441471 14843 ordered_multi_queue.h:149] Check failed: last_dispatched_key_ <= next_key_value_pair->sort_key (630822897702525020 vs. 630822897702522393) Non-sorted values added to queue: '(0, imu)'
**[FATAL] [946692970.445108370]: F0101 10:16:10.000000 14843 ordered_multi_queue.h:149] Check failed: last_dispatched_key_ <= next_key_value_pair->sort_key (630822897702525020 vs. 630822897702522393) Non-sorted values added to queue: '(0, imu)'**
*** Check failure stack trace: ***
    @ 0xb6a71060  (unknown)
    @ 0xb6a70f5c  (unknown)
    @ 0xb6a70b78  (unknown)
    @ 0xb6a72f98  (unknown)
    @    0xec18a  cartographer::common::OrderedMultiQueue<>::Dispatch()
    @    0xf4f68  cartographer::common::OrderedMultiQueue<>::Add()
    @    0xf507a  cartographer::mapping::SensorCollator<>::AddSensorData()
    @    0xf3e96  cartographer_ros::SensorDataProducer::AddLaserScanMessage()
    @    0xd8528  _ZN5boost6detail8function26void_function_obj_invoker1IZN16cartographer_ros12_GLOBAL__N_14Node10InitializeEvEUlRKNS_10shared_ptrIKN11sensor_msgs10LaserScan_ISaIvEEEEEE_vSE_E6invokeERNS1_15function_bufferESE_
    @    0xe05b0  boost::detail::function::void_function_obj_invoker1<>::invoke()
    @    0xeb204  ros::SubscriptionCallbackHelperT<>::call()
    @ 0xb6ccf580  (unknown)
    @ 0xb6ca0f12  (unknown)
    @ 0xb6ca1610  (unknown)
    @ 0xb6cd142e  (unknown)
    @ 0xb6cc15d4  (unknown)
    @    0xdacda  cartographer_ros::(anonymous namespace)::Run()
    @    0xd706c  main
    @ 0xb648f670  (unknown)
**[cartographer_node-14] process has died [pid 14843, exit code -6, cmd /home/ubuntu/google_cartographer/install_isolated/lib/cartographer_ros/cartographer_node -configuration_directory /home/ubuntu/google_cartographer/install_isolated/share/cartographer_turtlebot/configuration_files -configuration_basename turtlebot_rplidar_2d.lua points2:=/camera/depth/points __name:=cartographer_node __log:=/home/ubuntu/.ros/log/4cda3e64-bff1-11d3-9b70-7c5cf893e3bf/cartographer_node-14.log].
log file: /home/ubuntu/.ros/log/4cda3e64-bff1-11d3-9b70-7c5cf893e3bf/cartographer_node-14*.log**

At first it can run well, and I can recieve the map from the topic. But after a few seconds, it broke down. 
The lua file is here:
include "map_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  map_frame = "map",
  tracking_frame = "gyro_link",
  published_frame = "odom",
  odom_frame = "odom",
  provide_odom_frame = false,
  use_odometry_data = true,
  use_constant_odometry_variance = true,
  constant_odometry_translational_variance = 1e-7,
  constant_odometry_rotational_variance = 1e-7,
  use_horizontal_laser = true,
  use_horizontal_multi_echo_laser = false,
  horizontal_laser_min_range = 0.1,
  horizontal_laser_max_range = 6.,
  horizontal_laser_missing_echo_ray_length = 4.,
  num_lasers_3d = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
}

MAP_BUILDER.use_trajectory_builder_2d = true
TRAJECTORY_BUILDER_2D.use_imu_data = true
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
TRAJECTORY_BUILDER_2D.motion_filter.max_angle_radians = math.rad(0.1)

return options

hapi007, 2016-10-09:
i had met the same problem。

damonkohler, 2016-10-10:
I assume you're using the Kobuki platform. There's a driver bug that causes IMU messages to be published out of order. The flat_world_imu_node drops any packets that arrive out of order before they can reach the Cartographer node and cause a crash.

Did you modify the launch configuration? If so, perhaps you unintentionally bypassed the flat_world_imu_node.

Please share your launch file.

oya123, 2016-10-10:
@damonkohler  Yes, I'm using the Kobuki platform. I modified the launch configuration about the scan node, and I have updated the launch file in the issue.
Thanks a lot for your help.

damonkohler, 2016-10-10:
I'll close this then. @stonier, are you aware of this issue? There's a TODO in the source code :)

closed 2016-10-10 by damonkohler

stonier, 2016-10-10:
@damonkohler not aware of this one. Digging around the TODO's and hunting around the imu publishing code I can't find anything relevant nor anything untoward with the kobuki I have running here.

If you can point me in the right direction, I'll see it fixed.

damonkohler, 2016-10-10:
https://github.com/yujinrobot/kobuki/blob/kinetic/kobuki_node/src/library/slot_callbacks.cpp#L169

We dug into a bit before writing our flat_world_imu_node. It seems to us that a couple things can go wrong here: 

  * It's possible for the same IMU reading to be published more than once (see line 158).
  * Jitter in the sensor timing may result in line 172 computing something that's older than the last published data.

Hope that helps!

stonier, 2016-10-10:
Aye, it does. Thanks. Tracking the issue over [kobuki#370](https://github.com/yujinrobot/kobuki/issues/370)

ywj447, 2016-10-28:
Hi！@oya123  
I am trying to install cartographer on turtlebot(Kobuki) platform, and i get an error like this:

Building CXX object cartograp...nary.dir/lua_parameter_dictionary.cc.o
FAILED: /usr/bin/c++    -O3 -DNDEBUG -isystem /usr/local/include -isystem /usr/include/eigen3 -I. -I.. -I../cartographer/common/{Boost_INCLUDE_DIRS}     -O3 -DNDEBUG -std=c++11 -Wall -Wpedantic -Werror=format-security -Werror=return-type -Werror=uninitialized -MMD -MT cartographer/common/CMakeFiles/common_lua_parameter_dictionary.dir/lua_parameter_dictionary.cc.o -MF "cartographer/common/CMakeFiles/common_lua_parameter_dictionary.dir/lua_parameter_dictionary.cc.o.d" -o cartographer/common/CMakeFiles/common_lua_parameter_dictionary.dir/lua_parameter_dictionary.cc.o -c ../cartographer/common/lua_parameter_dictionary.cc
../cartographer/common/lua_parameter_dictionary.cc: In function 'void cartographer::common::{anonymous}::QuoteStringOnStack(lua_State*)':
../cartographer/common/lua_parameter_dictionary.cc:49:24: error: 'lua_pushglobaltable' was not declared in this scope
   lua_pushglobaltable(L);         // S: ... string globals

do you know how to solve it ? thanks a lot!

Milalaa, 2022-03-29:
@damonkohler Hi！I got the same error, but a little different. When I operate on turtlebot 2, everything works fine for mapping. But when I used the bag collected by turtlebot 2 to build a map, the following error occurred, and the map could not be displayed in rviz. Looking forward to your reply！

mina@ubuntu:~/carto_ws$ roslaunch cartographer_turtlebot turtlebot_rplidar_bag.launch 
... logging to /home/mina/.ros/log/04747788-af35-11ec-9ebf-000c29e037e4/roslaunch-ubuntu-77836.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:44389/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.12
 * /use_sim_time: True

NODES
  /
    cartographer_node (cartographer_ros/cartographer_node)
    cartographer_occupancy_grid_node (cartographer_ros/cartographer_occupancy_grid_node)
    flat_world_imu_node (cartographer_turtlebot/cartographer_flat_world_imu_node)
    playbag (rosbag/play)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [77846]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 04747788-af35-11ec-9ebf-000c29e037e4
process[rosout-1]: started with pid [77857]
started core service [/rosout]
process[cartographer_node-2]: started with pid [77860]
process[cartographer_occupancy_grid_node-3]: started with pid [77864]
process[flat_world_imu_node-4]: started with pid [77866]
process[rviz-5]: started with pid [77867]
process[playbag-6]: started with pid [77873]
[ INFO] [1648540279.291130994]: I0329 00:51:19.000000 77860 configuration_file_resolver.cc:41] Found '/home/mina/carto_ws/install_isolated/share/cartographer_turtlebot/configuration_files/turtlebot_urg_lidar_2d.lua' for 'turtlebot_urg_lidar_2d.lua'.
[ INFO] [1648540279.292502675]: I0329 00:51:19.000000 77860 configuration_file_resolver.cc:41] Found '/home/mina/carto_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1648540279.292568710]: I0329 00:51:19.000000 77860 configuration_file_resolver.cc:41] Found '/home/mina/carto_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1648540279.292623868]: I0329 00:51:19.000000 77860 configuration_file_resolver.cc:41] Found '/home/mina/carto_ws/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1648540279.292641974]: I0329 00:51:19.000000 77860 configuration_file_resolver.cc:41] Found '/home/mina/carto_ws/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1648540279.292747212]: I0329 00:51:19.000000 77860 configuration_file_resolver.cc:41] Found '/home/mina/carto_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1648540279.292772238]: I0329 00:51:19.000000 77860 configuration_file_resolver.cc:41] Found '/home/mina/carto_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1648540279.292808004]: I0329 00:51:19.000000 77860 configuration_file_resolver.cc:41] Found '/home/mina/carto_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1648540279.292823206]: I0329 00:51:19.000000 77860 configuration_file_resolver.cc:41] Found '/home/mina/carto_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1648540279.293074184]: I0329 00:51:19.000000 77860 configuration_file_resolver.cc:41] Found '/home/mina/carto_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1648540279.293135092]: I0329 00:51:19.000000 77860 configuration_file_resolver.cc:41] Found '/home/mina/carto_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1648540279.302026299]: I0329 00:51:19.000000 77860 map_builder_bridge.cc:135] Added trajectory with ID '0'.
[ WARN] [1648540280.036355120, 1648108208.752944214]: W0329 00:51:20.000000 77860 tf_bridge.cc:52] Lookup would require extrapolation at time 1648108208.665393500, but only time 1648108208.729968847 is in the buffer, when looking up transform from frame [laser] to frame [gyro_link]
[ INFO] [1648540280.095055691, 1648108208.814085611]: I0329 00:51:20.000000 77860 ordered_multi_queue.cc:172] All sensor data for trajectory 0 is available starting at '637837050087524683'.
[ INFO] [1648540280.165011192, 1648108208.875064462]: I0329 00:51:20.000000 77860 local_trajectory_builder_2d.cc:312] Extrapolator not yet initialized.
[ INFO] [1648540280.165120552, 1648108208.875064462]: I0329 00:51:20.000000 77860 local_trajectory_builder_2d.cc:135] Extrapolator is still initializing.
**F0329 00:51:20.165138 77860 ordered_multi_queue.cc:112] Check failed: last_dispatched_time_ <= next_data->GetTime() (637837050087725993 vs. 637837050087019812) Non-sorted data added to queue: '(0, imu)'
[FATAL] [1648540280.165331592, 1648108208.875064462]: F0329 00:51:20.000000 77860 ordered_multi_queue.cc:112] Check failed: last_dispatched_time_ <= next_data->GetTime() (637837050087725993 vs. 637837050087019812) Non-sorted data added to queue: '(0, imu)'**
*** Check failure stack trace: ***
[... cut ...]
TRAJECTORY_BUILDER_2D.max_range = 8.
TRAJECTORY_BUILDER_2D.missing_data_ray_length = 5.
TRAJECTORY_BUILDER_2D.use_imu_data = true
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
TRAJECTORY_BUILDER_2D.motion_filter.max_angle_radians = math.rad(0.1)

POSE_GRAPH.constraint_builder.min_score = 0.65
POSE_GRAPH.constraint_builder.global_localization_min_score = 0.7

return options
 ----------------------------------------------------------------------------------------------------------------------------
Thanks a lot！