cartographer-project/cartographer_turtlebot #24: About the flat_world_imu_node_main node ,maybe it has  a bug in line 52

hapi007, 2016-10-11:
i am using the Kobuki platform. I know that there's a driver bug that causes IMU messages to be published out of order. The flat_world_imu_node drops any packets that arrive out of order before they can reach the Cartographer node and cause a crash.
but  if  an imu data sequence is :header.stamp =10, header.stamp = 5, header.stamp =6,  header.stamp =11......, The  Cartographer node  will receive the data sequence like: 10, 6 ,11,..........  , it will also cause a crash  . Cartographer node  only want to received data:10 ,11 .........
So, i think that  the line 52 may be need to rewrite into the " if ' scope.  But, it will cause another question: if i change the system time  manually or using  NTP, it may drop too many packets.

damonkohler, 2016-10-11:
Definitely looks like a bug, thanks! Pushing a fix.

closed 2016-10-11 by damonkohler

hapi007, 2016-10-11:
thanks!  However,how about the next question? i mean if i change the system time manually or using NTP, e.g. from 10:00 to 9:00, the imu stamp will always out of order ,   does it will drop too many packets?

hapi007, 2016-10-11:
thanks! However, how about the next question? i mean if i change the system time manually or using NTP,e.g. from 10:00 clock to 09:00, the imu stamp will always out of order in this  hour, does it cause dropping too many packets?

SirVer, 2016-10-11:
If you change the system time while your ROS system is running you will have a bad time - this node will not behave nicely but most others will not either. Cartographer SLAM will also crash since it expects each sensor stream to arrive time ordered. You should not change the system time of a running system.

hapi007, 2016-10-11:
thanks a lot! SirVer

commit 33953ad4 (2016-10-11): Fixes #24. (#25)