cartographer-project/cartographer_turtlebot #37: Rosdep install failed

LucFabresse, 2016-12-07:
Hi,

I followed the instructions in the [doc](https://google-cartographer-ros-for-turtlebots.readthedocs.io/en/latest/) to install cartographer_turtlebot on an unbuntu 16.04 + ROS Kinetic. Everything is fine until I get a rosdep error during the install:

```
$ echo ${ROS_DISTRO}
kinetic
$ rosdep install --from-paths src --ignore-src --rosdistro=${ROS_DISTRO} -y

ERROR: Rosdep experienced an error: rosdep data for [type] must be a dictionary
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.5

Traceback (most recent call last): [...]
```

I tried on fresh 16.04+ROS Kinetic install and a fresh 14.04+ROS Indigo install and I got the same error.

Any hint?

Thanks,

SirVer, 2016-12-07:
​This sounds like some problem with rosdep. Did you file a bug? ​

Google Germany GmbH
Erika-Mann-Straße 33
80331 München

Registergericht und -nummer: Hamburg, HRB 86891
Sitz der Gesellschaft: Hamburg
Geschäftsführer: Matthew Scott Sucherman, Paul Terence Manicle

LucFabresse, 2016-12-07:
I did it here. Where else should I?

SirVer, 2016-12-07:
Well, I suggest you do what the output says you to do:

> Please go to the rosdep page [1] and file a bug report with the stack trace below.
> [1] : http://www.ros.org/wiki/rosdep

We can unfortunately not help with this, it is not a Cartographer, but a rosdep issue.

closed 2016-12-07 by SirVer

LucFabresse, 2016-12-07:
ok, I will.
Since I tried on a fresh install, I was more suspecting a problem in the way "cartographer" declares rosdep dependencies. But, I am probably wrong.
Thanks