cartographer-project/cartographer_turtlebot #44: Rosinstall file missing ceres-solver dependency

alexbaucom17, 2017-02-14:
Following the instructions exactly on the Read the Docs site causes rosdep install to fail as it can't resolve ceres-solver. I found that using wstool to merge the rosinstall file from cartograper_ros resolved this as that file does include the ceres-solver dependency. I think either updating the instructions to include this step or updating the rosinstall would fix this issue for future users.

renweizhukov, 2017-02-17:
I am hitting the same issue. rosdep install can't resolve ceres-solver.

```bash
$ rosdep install --from-paths src --ignore-src --rosdistro=${ROS_DISTRO} -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
cartographer: Cannot locate rosdep definition for [ceres-solver]
```

Also catkin build fails.

```bash
CMake Error at CMakeLists.txt:30 (find_package):
  By not providing "FindCeres.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Ceres", but
  CMake did not find one.

  Could not find a package configuration file provided by "Ceres" with any of
  the following names:

    CeresConfig.cmake
    ceres-config.cmake

  Add the installation prefix of "Ceres" to CMAKE_PREFIX_PATH or set
  "Ceres_DIR" to a directory containing one of the above files.  If "Ceres"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/renwei/workspace/ros/slam/google_cartographer_turtlebot/build_isolated/cartographer/install/CMakeFiles/CMakeOutput.log".
See also "/home/renwei/workspace/ros/slam/google_cartographer_turtlebot/build_isolated/cartographer/install/CMakeFiles/CMakeError.log".
<== Failed to process package 'cartographer': 
  Command '['cmake', '/home/renwei/workspace/ros/slam/google_cartographer_turtlebot/src/cartographer', '-DCMAKE_INSTALL_PREFIX=/home/renwei/workspace/ros/slam/google_cartographer_turtlebot/install_isolated', '-G', 'Ninja']' returned non-zero exit status 1
```

alexbaucom17, 2017-02-17:
You should be able to fix your problem with
```
wstool merge -t src https://raw.githubusercontent.com/googlecartographer/cartographer_ros/master/cartographer_ros.rosinstall
wstool update -t src
```
and then re-doing the rosdep install and continuing from there

renweizhukov, 2017-02-17:
Thank you for the workaround. Yeah, it works. But I am thinking why we can't add the dependency ceres-solver in the .rosinstall file of cartographer_turtlebot. I have created a pull request #45.

PR #46 (2017-02-21): Add the dependency ceres-solver in the .rosinstall file.
This commit fixes #44.

Change-Id: Idfb989aa23df74a37ddbe800e01ea366a0215b91
closed 2017-02-21 by SirVer