cartographer-project/cartographer_turtlebot #47: turtlebot_gazebo + cartographer will show error

stevenyslin, 2017-02-22:
Hello,
I have a problem when starting cartographer_turtlebot,

**Question:**
Since I don't have a real turtlebot, so I want to use turtlebot simulator to see how the result of cartographer,
my steps as following:
1. $ roslaunch turtlebot_gazebo turtlebot_world.launch
2. $ roslaunch cartographer_turtlebot demo_depth_camera_2d_v1.launch

and the **demo_depth_camera_2d_v1.launch** launch file as following:
I just modify this line `<remap from="imu_in" to="/mobile_base/sensors/imu_data" />`
```
<launch>

  <param name="/use_sim_time" value="true" />

  <node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
          -configuration_directory $(find cartographer_turtlebot)/configuration_files
          -configuration_basename turtlebot_depth_camera_2d.lua"
      output="screen" />

  <node name="flat_world_imu_node" pkg="cartographer_turtlebot"
      type="cartographer_flat_world_imu_node" output="screen">
    <remap from="imu_in" to="/mobile_base/sensors/imu_data" />
    <remap from="imu_out" to="/imu" />
  </node>

  <node name="squash_depth_camera" pkg="depthimage_to_laserscan"
      type="depthimage_to_laserscan">
    <remap from="image" to="/camera/depth/image_raw" />
  </node>

  <node name="rviz" pkg="rviz" type="rviz" required="true" args="-d $(find cartographer_turtlebot)/configuration_files/demo_turtlebot.rviz" />

</launch>
```

when I enter the two commands, it will show error message as following:
```
[ WARN] [1487748827.862433977, 4.500000000]: W0222 15:33:47.000000 19967 tf_bridge.cc:51] Lookup would require extrapolation into the past.  Requested time 4.240000000 but the earliest data is at time 4.570000000, when looking up transform from frame [camera_depth_frame] to frame [gyro_link]
[ WARN] [1487748827.862688735, 4.500000000]: W0222 15:33:47.000000 19967 tf_bridge.cc:51] Lookup would require extrapolation into the past.  Requested time 4.490000000 but the earliest data is at time 4.570000000, when looking up transform from frame [base_footprint] to frame [gyro_link]
F0222 15:33:47.862857 19967 sensor_bridge.cc:33] Check failed: frame_id[0] != '/' ('/' vs. '/') 
[FATAL] [1487748827.863191024, 4.500000000]: F0222 15:33:47.000000 19967 sensor_bridge.cc:33] Check failed: frame_id[0] != '/' ('/' vs. '/') 
*** Check failure stack trace: ***
    @     0x7f326c4b9daa  (unknown)
    @     0x7f326c4b9ce4  (unknown)
    @     0x7f326c4b96e6  (unknown)
    @     0x7f326c4bc687  (unknown)
    @           0x500a63  cartographer_ros::(anonymous namespace)::CheckNoLeadingSlash()
    @           0x500a95  cartographer_ros::SensorBridge::HandleRangefinder()
    @           0x501922  cartographer_ros::SensorBridge::HandleLaserScanMessage()
    @           0x4eb4dc  _ZN5boost6detail8function26void_function_obj_invoker1IZN16cartographer_ros12_GLOBAL__N_13RunEvEUlRKNS_10shared_ptrIKN11sensor_msgs10LaserScan_ISaIvEEEEEE_vSD_E6invokeERNS1_15function_bufferESD_
    @           0x4f0aab  boost::detail::function::void_function_obj_invoker1<>::invoke()
    @           0x4f7778  ros::SubscriptionCallbackHelperT<>::call()
    @     0x7f32693ac6b5  (unknown)
    @     0x7f3269366107  (unknown)
    @     0x7f3269366c33  (unknown)
    @     0x7f32693af1e5  (unknown)
    @     0x7f3269396e0b  (unknown)
    @           0x4ec8eb  cartographer_ros::(anonymous namespace)::Run()
    @           0x4e9d1d  main
    @     0x7f3268169f45  (unknown)
    @           0x4eb3c7  (unknown)
    @              (nil)  (unknown)
[cartographer_node-1] process has died [pid 19967, exit code -6, cmd /root/SLAM_project/googleSlam_ws/install_isolated/lib/cartographer_ros/cartographer_node -configuration_directory /root/SLAM_project/googleSlam_ws/install_isolated/share/cartographer_turtlebot/configuration_files -configuration_basename turtlebot_depth_camera_2d.lua __name:=cartographer_node __log:=/root/.ros/log/3b23272c-f8d1-11e6-9b7f-c86000043788/cartographer_node-1.log].
log file: /root/.ros/log/3b23272c-f8d1-11e6-9b7f-c86000043788/cartographer_node-1*.log

```

Thank you for your help.

SirVer, 2017-02-22:
The problem is here:

    Check failed: frame_id[0] != '/' ('/' vs. '/') 

One of your frames start with a '/'. This is deprecated behavior in ROS and we do not handle it. Please change your frames to not start with a slash.

closed 2017-02-22 by SirVer

stevenyslin, 2017-02-23:
Dear SirVer,

Really thanks a lot.
Here is my new turtlebot_world.launch.
```
<launch>

<!-- gazebo world -->
  <arg name="world_file"  default="$(env TURTLEBOT_GAZEBO_WORLD_FILE)"/>

  <arg name="base"      value="$(optenv TURTLEBOT_BASE kobuki)"/> <!-- create, roomba -->
  <arg name="battery"   value="$(optenv TURTLEBOT_BATTERY /proc/acpi/battery/BAT0)"/>  <!-- /proc/acpi/battery/BAT0 --> 
  <arg name="gui" default="true"/>
  <arg name="stacks"    value="$(optenv TURTLEBOT_STACKS hexagons)"/>  <!-- circles, hexagons --> 
  <arg name="3d_sensor" value="$(optenv TURTLEBOT_3D_SENSOR kinect)"/>  <!-- kinect, asus_xtion_pro --> 

  <include file="$(find gazebo_ros)/launch/empty_world.launch">
    <arg name="use_sim_time" value="true"/>
    <arg name="debug" value="false"/>
    <arg name="gui" value="$(arg gui)" />
    <arg name="world_name" value="$(arg world_file)"/>
  </include>
  
  <include file="$(find turtlebot_gazebo)/launch/includes/$(arg base).launch.xml">
    <arg name="base" value="$(arg base)"/>
    <arg name="stacks" value="$(arg stacks)"/>
    <arg name="3d_sensor" value="$(arg 3d_sensor)"/>
  </include>
  
  <node pkg="robot_state_publisher" type="robot_state_publisher" name="robot_state_publisher">
    <param name="publish_frequency" type="double" value="30.0" />
  </node>
  
  <!-- Fake laser -->
  <node pkg="nodelet" type="nodelet" name="laserscan_nodelet_manager" args="manager"/>
  <node pkg="nodelet" type="nodelet" name="depthimage_to_laserscan"
        args="load depthimage_to_laserscan/DepthImageToLaserScanNodelet laserscan_nodelet_manager">
    <param name="scan_height" value="10"/>
    <!-- change start -->
    <param name="output_frame_id" value="camera_depth_frame"/>**
    <param name="range_min" value="0.45"/>
    <remap from="image" to="camera/depth/image_raw"/>
    <remap from="scan" to="scan"/>
    <!-- change end -->
  </node>

  <node pkg="tf" type="static_transform_publisher" name="odom_map_broadcaster" args="0 0 0 0 0 0 /map /odom 100" />

</launch>
```

stevenyslin, 2017-02-23:
And I have another question after I enter this command ($ roslaunch cartographer_turtlebot demo_depth_camera_2d_v1.launch ).

```
SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.20
 * /use_sim_time: True

NODES
  /
    cartographer_node (cartographer_ros/cartographer_node)
    flat_world_imu_node (cartographer_turtlebot/cartographer_flat_world_imu_node)
    rviz (rviz/rviz)
    squash_depth_camera (depthimage_to_laserscan/depthimage_to_laserscan)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[cartographer_node-1]: started with pid [32478]
process[flat_world_imu_node-2]: started with pid [32479]
[ INFO] [1487835085.640725607]: I0223 15:31:25.000000 32478 configuration_file_resolver.cc:40] Found '/root/SLAM_project/googleSlam_ws/install_isolated/share/cartographer_turtlebot/configuration_files/turtlebot_depth_camera_2d.lua' for 'turtlebot_depth_camera_2d.lua'.
[ INFO] [1487835085.641221913]: I0223 15:31:25.000000 32478 configuration_file_resolver.cc:40] Found '/root/SLAM_project/googleSlam_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1487835085.641301836]: I0223 15:31:25.000000 32478 configuration_file_resolver.cc:40] Found '/root/SLAM_project/googleSlam_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1487835085.641428447]: I0223 15:31:25.000000 32478 configuration_file_resolver.cc:40] Found '/root/SLAM_project/googleSlam_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1487835085.641518760]: I0223 15:31:25.000000 32478 configuration_file_resolver.cc:40] Found '/root/SLAM_project/googleSlam_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1487835085.641771008]: I0223 15:31:25.000000 32478 configuration_file_resolver.cc:40] Found '/root/SLAM_project/googleSlam_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1487835085.641838761]: I0223 15:31:25.000000 32478 configuration_file_resolver.cc:40] Found '/root/SLAM_project/googleSlam_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1487835085.642015666]: I0223 15:31:25.000000 32478 configuration_file_resolver.cc:40] Found '/root/SLAM_project/googleSlam_ws/install_isolated/share/cartographer/configuration_files/sparse_pose_graph.lua' for 'sparse_pose_graph.lua'.
[ INFO] [1487835085.642069513]: I0223 15:31:25.000000 32478 configuration_file_resolver.cc:40] Found '/root/SLAM_project/googleSlam_ws/install_isolated/share/cartographer/configuration_files/sparse_pose_graph.lua' for 'sparse_pose_graph.lua'.
process[squash_depth_camera-3]: started with pid [32561]
process[rviz-4]: started with pid [32562]
[ INFO] [1487835085.697343493]: I0223 15:31:25.000000 32478 submaps.cc:176] Added submap 1
[ INFO] [1487835085.697462334]: I0223 15:31:25.000000 32478 map_builder_bridge.cc:37] Added trajectory with ID '0'.
[ WARN] [1487835085.943039301, 9.150000000]: W0223 15:31:25.000000 32478 tf_bridge.cc:51] Lookup would require extrapolation at time 9.120000000, but only time 9.640000000 is in the buffer, when looking up transform from frame [base_footprint] to frame [gyro_link]
[ WARN] [1487835085.943278409, 9.150000000]: W0223 15:31:25.000000 32478 tf_bridge.cc:51] Lookup would require extrapolation at time 9.120000000, but only time 9.640000000 is in the buffer, when looking up transform from frame [base_link] to frame [gyro_link]
[ WARN] [1487835085.943408195, 9.150000000]: W0223 15:31:25.000000 32478 tf_bridge.cc:51] Lookup would require extrapolation at time 9.130000000, but only time 9.640000000 is in the buffer, when looking up transform from frame [base_footprint] to frame [gyro_link]
[ WARN] [1487835085.943519530, 9.150000000]: W0223 15:31:25.000000 32478 tf_bridge.cc:51] Lookup would require extrapolation at time 9.130000000, but only time 9.640000000 is in the buffer, when looking up transform from frame [base_link] to frame [gyro_link]
[ WARN] [1487835085.943793722, 9.150000000]: W0223 15:31:25.000000 32478 tf_bridge.cc:51] Lookup would require extrapolation at time 9.140000000, but only time 9.640000000 is in the buffer, when looking up transform from frame [base_footprint] to frame [gyro_link]
[ WARN] [1487835085.943960775, 9.150000000]: W0223 15:31:25.000000 32478 tf_bridge.cc:51] Lookup would require extrapolation at time 9.140000000, but only time 9.640000000 is in the buffer, when looking up transform from frame [base_link] to frame [gyro_link]
[ WARN] [1487835085.951152242, 9.160000000]: W0223 15:31:25.000000 32478 tf_bridge.cc:51] Lookup would require extrapolation at time 9.150000000, but only time 9.640000000 is in the buffer, when looking up transform from frame [base_footprint] to frame [gyro_link]
[ WARN] [1487835085.951273443, 9.160000000]: W0223 15:31:25.000000 32478 tf_bridge.cc:51] Lookup would require extrapolation at time 9.150000000, but only time 9.640000000 is in the buffer, when looking up transform from frame [base_link] to frame [gyro_link]
[ WARN] [1487835085.961300095, 9.170000000]: W0223 15:31:25.000000 32478 tf_bridge.cc:51] Lookup would require extrapolation at time 9.160000000, but only time 9.640000000 is in the buffer, when looking up transform f [... cut ...] _depth_camera_2d.lua __name:=cartographer_node __log:=/root/.ros/log/0e2a3fc0-f99a-11e6-9b7f-c86000043788/cartographer_node-1.log].
log file: /root/.ros/log/0e2a3fc0-f99a-11e6-9b7f-c86000043788/cartographer_node-1*.log
```

And I use tool to trace error

**rqt_graph:**
http://imgur.com/a/orRqW

**frames tree:**
http://imgur.com/a/b8ZkN

It seems like the **flat_world_imu_node** can't output **/imu** frame, and the cartographer_node don't work.
Please help to solve the problem, thank you so much.

SirVer, 2017-02-23:
The error is this:

    Check failed: sensor_to_tracking->translation().norm() < 1e-5 The IMU frame must be colocated with the tracking frame.

Your IMU is not at the same origin as the tracking frame, i.e. the frame you want to track with SLAM. This is not supported since angular velocities cannot be transformed out of this mathematically correct. So you always have to track your IMU (though rotations are okay).

stevenyslin, 2017-02-23:
Dear SirVer,

When I modify the **turtlebot_depth_camera_2d.lua** file, the red error will disappear.
I modify this line:
`tracking_frame = "mobile_base/sensors/imu_data",`

But there has a new warning on my terminal
```
[ WARN] [1487841837.999211894, 78.940000000]: W0223 17:23:57.000000 20101 tf_bridge.cc:51] "mobile_base/sensors/imu_data" passed to lookupTransform argument target_frame does not exist.
```
May I ask if this is correct? Thank you.

SirVer, 2017-02-23:
> May I ask if this is correct? 

Unlikely. This looks like a topic name, not like a frame name. Use `rostopic echo -n1 <topic>` to figure out the frame_id of this.

stevenyslin, 2017-02-23:
Dear SirVer,

Thank you for your prompt reply.
I enter this command `rostopic echo -n1 /mobile_base/sensors/imu_data`, and the result as following:
```
header: 
  seq: 23485
  stamp: 
    secs: 234
    nsecs: 860000000
  frame_id: base_link
orientation: 
  x: -0.000250574373646
  y: -0.00399240017799
  z: -0.0692090905304
  w: 0.997594155828
orientation_covariance: [1000000.0, 0.0, 0.0, 0.0, 1000000.0, 0.0, 0.0, 0.0, 0.05]
angular_velocity: 
  x: 8.99941052281e-05
  y: 0.000162501806715
  z: -0.000645573780137
angular_velocity_covariance: [1000000.0, 0.0, 0.0, 0.0, 1000000.0, 0.0, 0.0, 0.0, 0.05]
linear_acceleration: 
  x: -0.00877279334463
  y: -0.0958036597477
  z: 9.73305207043
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
```
And I modify again this line,
`tracking_frame = "base_link",`

After enter command, it shows a lot of error message,
```
SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.20
 * /use_sim_time: True

NODES
  /
    cartographer_node (cartographer_ros/cartographer_node)
    flat_world_imu_node (cartographer_turtlebot/cartographer_flat_world_imu_node)
    rviz (rviz/rviz)
    squash_depth_camera (depthimage_to_laserscan/depthimage_to_laserscan)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[cartographer_node-1]: started with pid [24010]
process[flat_world_imu_node-2]: started with pid [24011]
[ INFO] [1487845529.215956672]: I0223 18:25:29.000000 24010 configuration_file_resolver.cc:40] Found '/root/SLAM_project/googleSlam_ws/install_isolated/share/cartographer_turtlebot/configuration_files/turtlebot_depth_camera_2d.lua' for 'turtlebot_depth_camera_2d.lua'.
[ INFO] [1487845529.216333076]: I0223 18:25:29.000000 24010 configuration_file_resolver.cc:40] Found '/root/SLAM_project/googleSlam_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1487845529.216384575]: I0223 18:25:29.000000 24010 configuration_file_resolver.cc:40] Found '/root/SLAM_project/googleSlam_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1487845529.216491093]: I0223 18:25:29.000000 24010 configuration_file_resolver.cc:40] Found '/root/SLAM_project/googleSlam_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1487845529.216539520]: I0223 18:25:29.000000 24010 configuration_file_resolver.cc:40] Found '/root/SLAM_project/googleSlam_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1487845529.216656572]: I0223 18:25:29.000000 24010 configuration_file_resolver.cc:40] Found '/root/SLAM_project/googleSlam_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1487845529.216700474]: I0223 18:25:29.000000 24010 configuration_file_resolver.cc:40] Found '/root/SLAM_project/googleSlam_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1487845529.216840497]: I0223 18:25:29.000000 24010 configuration_file_resolver.cc:40] Found '/root/SLAM_project/googleSlam_ws/install_isolated/share/cartographer/configuration_files/sparse_pose_graph.lua' for 'sparse_pose_graph.lua'.
[ INFO] [1487845529.216881569]: I0223 18:25:29.000000 24010 configuration_file_resolver.cc:40] Found '/root/SLAM_project/googleSlam_ws/install_isolated/share/cartographer/configuration_files/sparse_pose_graph.lua' for 'sparse_pose_graph.lua'.
process[squash_depth_camera-3]: started with pid [24093]
[ INFO] [1487845529.272803408]: I0223 18:25:29.000000 24010 submaps.cc:176] Added submap 1
[ INFO] [1487845529.272903709]: I0223 18:25:29.000000 24010 map_builder_bridge.cc:37] Added trajectory with ID '0'.
process[rviz-4]: started with pid [24094]
[ WARN] [1487845529.438571720, 9.890000000]: W0223 18:25:29.000000 24010 tf_bridge.cc:51] "base_link" passed to lookupTransform argument target_frame does not exist. 
[ WARN] [1487845529.448378397, 9.900000000]: W0223 18:25:29.000000 24010 tf_bridge.cc:51] Lookup would require extrapolation at time 9.880000000, but only time 10.400000000 is in the buffer, when looking up transform from frame [base_footprint] to frame [base_link]
[ WARN] [1487845529.448539830, 9.900000000]: W0223 18:25:29.000000 24010 tf_bridge.cc:51] Lookup would require extrapolation at time 9.890000000, but only time 10.400000000 is in the buffer, when looking up transform from frame [base_footprint] to frame [base_link]
[ WARN] [1487845529.458250830, 9.910000000]: W0223 18:25:29.000000 24010 tf_bridge.cc:51] Lookup would require extrapolation at time 9.900000000, but only time 10.400000000 is in the buffer, when looking up transform from frame [base_footprint] to frame [base_link]
[... 93 lines cut ...]
    @           0x5dc6bb  _ZNSt17_Function_handlerIFvSt10unique_ptrIN12cartographer6sensor4DataESt14default_deleteIS3_EEEZNS2_8Collator13AddTrajectoryEiRKSt13unordered_setISsSt4hashISsESt8equal_toISsESaISsEESt8functionIFvRKSsS6_EEEUlS6_E_E9_M_invokeERKSt9_Any_dataS6_
    @           0x5dfa03  std::function<>::operator()()
    @           0x5dea5f  cartographer::sensor::OrderedMultiQueue::Dispatch()
    @           0x5df482  cartographer::sensor::OrderedMultiQueue::Add()
    @           0x5dc8de  cartographer::sensor::Collator::AddSensorData()
[... cut ...]
[cartographer_node-1] process has died [pid 24010, exit code -6, cmd /root/SLAM_project/googleSlam_ws/install_isolated/lib/cartographer_ros/cartographer_node -configuration_directory /root/SLAM_project/googleSlam_ws/install_isolated/share/cartographer_turtlebot/configuration_files -configuration_basename turtlebot_depth_camera_2d.lua __name:=cartographer_node __log:=/root/.ros/log/5e95a6f8-f9b2-11e6-9b7f-c86000043788/cartographer_node-1.log].
log file: /root/.ros/log/5e95a6f8-f9b2-11e6-9b7f-c86000043788/cartographer_node-1*.log

```

Thank you for your help.

SirVer, 2017-02-23:
Seems like you feed Cartographer wrong data. Please provide configuration (in a git branch based on cartographre_ros) & bag - otherwise it is impossible to say what is wrong.

stevenyslin, 2017-02-24:
Dear SirVer,

Is there an example for configuration that I can provide.
And I'm not using rosbag to play bag, I use this command to simulate turtlebot `roslaunch turtlebot_gazebo turtlebot_world.launch`

And I want ask another question:
Does cartographer_turtlebot can update rviz map, when a object is moving?
For example, when the toyota HSR detect four chair at **0:57**, and update the map at **1:15**.
https://www.youtube.com/watch?v=cK6s7soVwws&list=LL_Uuv-CU37JaR2XiILvtBvQ#t=57s

Thank you so much.

SirVer, 2017-02-24:
> Does cartographer_turtlebot can update rviz map, when a object is moving?

This happens automatically due to the nature of the probabilistic mapping.

Unfortunately I am not familiar with the gazebo integration of turtlebot. It might be that there is some duplication in this ros package and ours. Use roswtf to figure out issues, look at the rViz info for your individual sensor positions.

stevenyslin, 2017-02-24:
Dear SirVer,

I got three error as following:
```
Loaded plugin tf.tfwtf
No package or stack in context
================================================================================
Static checks summary:

No errors or warnings
================================================================================
Beginning tests of your ROS graph. These may take awhile...
analyzing graph...
... done analyzing graph
running graph rules...
ERROR: connection refused to [http://steven-U46A:46170/]
... done running graph rules
running tf checks, this will take a second...
... tf checks complete

Online checks summary:

Found 2 warning(s).
Warnings are things that may be just fine, but are sometimes at fault

WARNING The following node subscriptions are unconnected:
 * /mobile_base_nodelet_manager:
   * /cmd_vel_mux/input/switch
   * /mobile_base/sensors/core
   * /cmd_vel_mux/input/navi
   * /cmd_vel_mux/input/safety_controller
   * /cmd_vel_mux/input/teleop
 * /gazebo:
   * /gazebo/set_link_state
   * /mobile_base/commands/reset_odometry
   * /gazebo/set_model_state
   * /mobile_base/commands/motor_power

WARNING These nodes have died:
 * cartographer_node-1
 * spawn_turtlebot_model-4

Found 3 error(s).

ERROR Could not contact the following nodes:
 * /cartographer_node

ERROR The following nodes should be connected but aren't:
 * /laserscan_nodelet_manager->/cartographer_node (/scan)
 * /gazebo->/cartographer_node (/tf)
 * /flat_world_imu_node->/cartographer_node (/imu)
 * /gazebo->/cartographer_node (/clock)
 * /squash_depth_camera->/cartographer_node (/scan)
 * /odom_map_broadcaster->/cartographer_node (/tf)
 * /robot_state_publisher->/cartographer_node (/tf)
 * /cartographer_node->/cartographer_node (/tf)
 * /gazebo->/cartographer_node (/odom)

ERROR Errors connecting to the following services:
 * service [/finish_trajectory] appears to be malfunctioning: Unable to communicate with service [/finish_trajectory], address [rosrpc://steven-U46A:44329]
 * service [/cartographer_node/get_loggers] appears to be malfunctioning: Unable to communicate with service [/cartographer_node/get_loggers], address [rosrpc://steven-U46A:44329]
 * service [/cartographer_node/set_logger_level] appears to be malfunctioning: Unable to communicate with service [/cartographer_node/set_logger_level], address [rosrpc://steven-U46A:44329]
 * service [/submap_query] appears to be malfunctioning: Unable to communicate with service [/submap_query], address [rosrpc://steven-U46A:44329]
```

And rviz didn't show any error.
http://imgur.com/a/0ePRM

Really thank you for your help.

SirVer, 2017-02-24:
It seems like your node wiring is not correct - i.e. a ROS misconfiguration. I do not think this is Cartographer related , maybe you can ask at http://answers.ros.org/questions/ for ROS instructions.

mojovski, 2018-05-04:
Same here.
I assume the problem is that the xarco files in Gazebo for Turtlebo3 are not valid from cartographers perspective.
@SirVer your wrote:
> Your IMU is not at the same origin as the tracking frame, i.e. the frame you want to track with SLAM. This is not supported since angular velocities cannot be transformed out of this mathematically correct. So you always have to track your IMU (though rotations are okay).

So currently I am using this urdf.xarco file for the turtlebot:
https://github.com/ROBOTIS-GIT/turtlebot3/blob/master/turtlebot3_description/urdf/turtlebot3_waffle.urdf.xacro#L148

I dont quite understand, the meaning of your statement. 
* Gazebo sends imu readings in the topic /imu.
* these are located in the frame imu_link (? if not, how do I know?)

Would be really great if you find time to explain it a little more. 

Thanks alot!

BTW: The Lua-Config file contains:
```lua
map_frame = "map", --    The ROS frame ID to use for publishing submaps, the parent frame of poses, usually “map”.
  tracking_frame = "imu_link", --The ROS frame ID of the frame that is tracked by the SLAM algorithm. If an IMU is used, it should be at its position, although it might be rotated. A common choice is “imu_link”.
  published_frame = "base_link", -- The ROS frame ID to use as the child frame for publishing poses. For example “odom” if an “odom” frame is supplied by a different part of the system. In this case the pose of “odom” in the map_frame will be published. Otherwise, setting it to “base_link” is likely appropriate.
  odom_frame = "odom", --Only used if provide_odom_frame is true. The frame between published_frame and map_frame to be used for publishing the (non-loop-closed) local SLAM result. Usually “odom”.
  
```