cartographer-project/cartographer_turtlebot #52: Plan for Stable Release?

alexbaucom17, 2017-05-15:
Is there any plan for a stable release which we can reference in separate projects? 

My team recently released some code which used Cartographer Turtlebot with a custom parameter file and instructed users to install the Cartographer code via the read the docs site. However, shortly after this we discovered that the default parameter file had been changed and since new users were pulling the most recent code, they got all sorts of errors when using the parameter file we provided. 

This isn't the end of the world as it ended up being a pretty simple fix but I was just curious if there was any plans to publish a static/stable release that  others can reference so we don't run into this issue again? Given the current installation instructions, would the only real option be to tell our users to checkout a specific commit from each cartographer repository before proceeding with the installation?

SirVer, 2017-05-16:
Cartographer is pre-1.0 as of now. We plan to follow semantic versioning eventually, but as of now do not have a stable public API.

We plan to tag a 0.1.0 pre-releases soon, but we give no stability guarantees until 1.0.

alexbaucom17, 2017-05-16:
No problem. I figured you guys were still in the early stages. I'll keep an eye out for the pre-releases and go from there. Thanks!

closed 2017-05-16 by alexbaucom17