cartographer-project/cartographer_turtlebot #53: Poor map quality using depth_camera_2d due to map TF jitter

mattcurfman, 2017-05-16:
I'd appreciate another set of eyes to see what I might be doing wrong?

I'm using the demo_depth_camera_2d.launch stock from this repo, using a bag captured using the Intel RealSense ZR300 camera (https://click.intel.com/intelr-realsensetm-development-kit-featuring-the-zr300.html) instead of the Asus Xtion Pro camera used in the reference bag.  

The map TF seems to randomly oscillate/jitter between a normal pose and something with the X and Y axis pointing somewhere else.  The map -> odom transform is thus affected causing a poor quality map.
 
I've provided a sample bag file @ https://drive.google.com/open?id=0B-PswUXGJd8uU0RLcTVGR3NTZzQ

The same bag file but using the demo_lidar_2d.launch and the rplidar topic stream instead of the depth camera topic stream works fine.  Any suggestions what is causing this behavior?

SirVer, 2017-05-17:
Could you also provide your configuration (if changed) and the comand lines you are using to launch your ROS environment.

mattcurfman, 2017-05-17:
For playback I'm using "roslaunch cartographer_turtlebot demo_depth_camera2d.launch bag_filename:=/home/user/test.bag" unchanged.  

For recording, I'm using a standard turtlebot 'minimal.launch', modified to use ZR300 model @ https://github.com/IntelRealSense/realsense_samples_ros/blob/kinetic-devel/realsense_ros_camera/urdf/_zr300.urdf.xacro mounted on polls 25cm above the top plate.

macmason, 2017-05-23:
Sorry to be slow getting back to you on this!

Here's what I think is going on: your RealSense is far, far too noisy for scan matching to really work, at least without clipping the range.

Here's how I debugged:

1. I started up `roscore` and `rviz` manually (set the fixed frame to `odom`).
2. I visualized the `/scan topic` (which gives a pretty good idea what's actually there).
3. I manually started up the `depthimage_to_laserscan` node that Cartographer uses to get 2D.
4. I visualized that, too, and ran your bag.

At ranges of more than a meter or two, there's serious, serious jitter in that cloud: scan-matching with it seems difficult, if not impossible. I think your next step is to drop all the points from that cloud that are beyond a certain distance (the magical number for a Kinect-like sensor is about 3m; I don't have a ballpark for a RealSense, but this visualization will probably give you a place to start.)

mattcurfman, 2017-05-25:
Thanks @macmason for the pointer.  Noise can be tuned, now that I know what too look at in more detail.
-Matt

macmason, 2017-05-25:
Happy to help!

Closing this.

closed 2017-05-25 by mattcurfman