cartographer-project/cartographer_turtlebot #75: Why my map is coming out so bad? Please help me with tuning!

Python43coder, 2017-11-06:
Hello,

I recently tried using **_google cartographer_** to generate a map of my building. I use to use _**gmapping**_ but i didn't get very good results, so i switched to gmapping.

When I did couple of small test runs I got good results, but when i did try mapping my whole building it didn't give me good results at all ( please see the map below), This is what I am using:

- Ubuntu 14.04

- ROS Indigo

- Turtlebot with Kinect 1.0 sensor

I looked into couple of other issues on tuning and tried to modify my config file to get good results, but i think i didn't do a great job.

Here is the map I got:

[https://drive.google.com/open?id=1rIExNvfOZo7HN-z88YTISyitbxZ_cphF](url)

Here are my config files:
[https://drive.google.com/open?id=1-4MlGU8nfoUDYCQ8lwcL4PwoPtBqP9tk](url)

[https://drive.google.com/open?id=1EZ-7aFo3_5lfCyYU8RK7T6BdQpBhnvAu](url)

here is my data bag:

[https://drive.google.com/open?id=1z1U_gHrWXjJeyE8cqV3IRHIalMb4c0D8](url)

Can anyone of you please help me?

gaschler, 2017-11-07:
Without having a deeper look yet, have you tried the depth camera configurations such as https://github.com/googlecartographer/cartographer_turtlebot/blob/master/cartographer_turtlebot/configuration_files/turtlebot_depth_camera_3d.lua ?

Python43coder, 2017-11-07:
Yes, I tried different parameters, but i didn't get good results.

SirVer, 2017-11-14:
Running `rosrun cartographer_ros cartographer_rosbag_validate -bag_filename $HOME/Downloads/full_B4U_carto.bag` shows that your gyro is jumping backwards in time, so you should probably not use the IMU. But you do not, which is good.

This whole bug reminds me a lot of #70 and it seems like you started from my configuration there. Unfortunately, your odometry data is also not super great, seemingly underestimating left turns systematically. If you could fix that you'd have a better chance of getting a good map - you could also try not always turning left, but also sometimes right (You only have one right turn in your bag). Later it seems like a wheel also starts slipping since the robot is driving right turns in my config.

Your laser also regularly does not sees anything but the walls to its side, which yields in many similar looking scans & submaps which result in many poor loop closure constraints being added.

I think fundamentally you'd require a longer range laser and ideally better odometry to get good data for your map. As it stands, Cartographer can barely do its job with this, sorry.

closed 2017-11-14 by SirVer