cartographer-project/cartographer_turtlebot #76: KinectV2 in cartographer 2D Map using turtlebot2(kobuki)

mozheng, 2017-11-10:
I'm a petty new person in ROS system.I did not write launch file ago. I want to building 2D-MAP using KinectV2 and turtlebot2.
This is my launch file. I cannot see anything by rviz. Please tell me where can I fix it. and, what should I do next. Thanks.

```
<?xml version="1.0"?>
<launch>    
    <!-- start kinect2-->            
    <include file="$(find kinect2_bridge)/launch/kinect2_bridge.launch">
        <arg name="base_name"         value="kinect2"/>
        <arg name="sensor"            value=""/>
        <arg name="publish_tf"        value="true"/>
        <arg name="base_name_tf"      value="kinect2"/>
        <arg name="fps_limit"         value="-1.0"/>
        <arg name="calib_path"        value="$(find kinect2_bridge)/data/"/>
        <arg name="use_png"           value="false"/>
        <arg name="jpeg_quality"      value="90"/>
        <arg name="png_level"         value="1"/>
        <arg name="depth_method"      value="default"/>
        <arg name="depth_device"      value="-1"/>
        <arg name="reg_method"        value="default"/>
        <arg name="reg_device"        value="-1"/>
        <arg name="max_depth"         value="12.0"/>
        <arg name="min_depth"         value="0.1"/>
        <arg name="queue_size"        value="5"/>
        <arg name="bilateral_filter"  value="true"/>
        <arg name="edge_aware_filter" value="true"/>
        <arg name="worker_threads"    value="4"/>
    </include>  
  <param name="/depthimage_to_laserscan/output_frame_id"
      value="camera_depth_frame" />

  <arg name="configuration_basename" value="turtlebot_depth_camera_2d.lua" />
  <include file="$(find turtlebot_bringup)/launch/minimal.launch" />

  <node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
          -configuration_directory
              $(find cartographer_turtlebot)/configuration_files
          -configuration_basename $(arg configuration_basename)"
      output="screen">
   <remap from="points2" to="/camera/depth_registered/points" />
  </node>

  <node name="flat_world_imu_node" pkg="cartographer_turtlebot"
      type="cartographer_flat_world_imu_node" output="screen">
    <remap from="imu_in" to="/mobile_base/sensors/imu_data_raw" />
    <remap from="imu_out" to="/imu" />
  </node>

  <node name="cartographer_occupancy_grid_node" pkg="cartographer_ros"
      type="cartographer_occupancy_grid_node" args="-resolution 0.05" />

SirVer, 2017-11-10:
Please provide full console output, commandline, bag and configuration files (.lua).

SirVer, 2017-11-14:
Closing for inactivity.

closed 2017-11-14 by SirVer