cartographer-project/cartographer_turtlebot #80: File 'trajectory_builder.lua' was not found.

FelicxFoster, 2017-12-08:
#why i have this error
i run the "roslaunch cartographer_ros turtlebot_lidar_2d.launch" the terminal place that "[FATAL] [1512705370.945497189]: F1208 11:56:10.000000 22035 configuration_file_resolver.cc:44] File 'trajectory_builder.lua' was not found."   How could i solve it, please help me!

felicx@felicx-Lenovo-XiaoXin-310-14ISK:~/carto_ws$ roslaunch cartographer_ros turtlebot_lidar_2d.launch
... logging to /home/felicx/.ros/log/5327b740-dbcd-11e7-b5a9-ac2b6e7a2f6f/roslaunch-felicx-Lenovo-XiaoXin-310-14ISK-23951.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://felicx-Lenovo-XiaoXin-310-14ISK:41487/

SUMMARY
========

PARAMETERS
 * /app_manager/auto_rapp_installation: False
 * /app_manager/auto_start_rapp: 
 * /app_manager/capability_server_name: capability_server
 * /app_manager/local_remote_controllers_only: False
 * /app_manager/preferred: [{'rocon_apps/chi...
 * /app_manager/rapp_package_blacklist: []
 * /app_manager/rapp_package_whitelist: ['rocon_apps', 't...
 * /app_manager/robot_icon: turtlebot_bringup...
 * /app_manager/robot_name: turtlebot
 * /app_manager/robot_type: turtlebot
 * /app_manager/screen: True
 * /app_manager/simulation: False
 * /app_manager/use_gateway_uuids: False
 * /bumper2pointcloud/pointcloud_radius: 0.24
 * /capability_server/blacklist: ['std_capabilitie...
 * /capability_server/defaults/kobuki_capabilities/KobukiBringup: kobuki_capabiliti...
 * /capability_server/defaults/kobuki_capabilities/KobukiBumper: kobuki_capabiliti...
 * /capability_server/defaults/kobuki_capabilities/KobukiCliffDetection: kobuki_capabiliti...
 * /capability_server/defaults/kobuki_capabilities/KobukiLED1: kobuki_capabiliti...
 * /capability_server/defaults/kobuki_capabilities/KobukiLED2: kobuki_capabiliti...
 * /capability_server/defaults/kobuki_capabilities/KobukiLED: kobuki_capabiliti...
 * /capability_server/defaults/kobuki_capabilities/KobukiWheelDropDetection: kobuki_capabiliti...
 * /capability_server/defaults/std_capabilities/Diagnostics: turtlebot_capabil...
 * /capability_server/defaults/std_capabilities/DifferentialMobileBase: kobuki_capabiliti...
 * /capability_server/defaults/std_capabilities/LaserSensor: turtlebot_capabil...
 * /capability_server/defaults/std_capabilities/RGBDSensor: turtlebot_capabil...
 * /capability_server/defaults/std_capabilities/RobotStatePublisher: turtlebot_capabil...
 * /capability_server/defaults/turtlebot_capabilities/TurtleBotBringup: turtlebot_capabil...
 * /capability_server/nodelet_manager_name: capability_server...
 * /capability_server/package_whitelist: ['kobuki_capabili...
 * /cmd_vel_mux/yaml_cfg_file: /home/felicx/turt...
 * /description: Kick-ass ROS turtle
 * /diagnostic_aggregator/analyzers/input_ports/contains: ['Digital Input',...
 * /diagnostic_aggregator/analyzers/input_ports/path: Input Ports
 * /diagnostic_aggregator/analyzers/input_ports/remove_prefix: mobile_base_nodel...
 * /diagnostic_aggregator/analyzers/input_ports/timeout: 5.0
 * /diagnostic_aggregator/analyzers/input_ports/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/kobuki/contains: ['Watchdog', 'Mot...
 * /diagnostic_aggregator/analyzers/kobuki/path: Kobuki
 * /diagnostic_aggregator/analyzers/kobuki/remove_prefix: mobile_base_nodel...
 * /diagnostic_aggregator/analyzers/kobuki/timeout: 5.0
 * /diagnostic_aggregator/analyzers/kobuki/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/power/contains: ['Battery', 'Lapt...
 * /diagnostic_aggregator/analyzers/power/path: Power System
 * /diagnostic_aggregator/analyzers/power/remove_prefix: mobile_base_nodel...
 * /diagnostic_aggregator/analyzers/power/timeout: 5.0
 * /diagnostic_aggregator/analyzers/power/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/sensors/contains: ['Cliff Sensor', ...
 * /diagnostic_aggregator/analyzers/sensors/path: Sensors
 * /diagnostic_aggregator/analyzers/sensors/remove_prefix: mobile_base_nodel...
 * /diagnostic_aggregator/analyzers/sensors/timeout: 5.0
 * /diagnostic_aggregator/analyzers/sensors/type: diagnostic_aggreg...
 * /diagnostic_aggregator/base_path: 
 * /diagnostic_aggregator/pub_rate: 1.0
 * /icon: turtlebot_bringup...
[... 110 lines cut ...]
[rviz-16] killing on exit
[zeroconf/zeroconf-13] killing on exit
[interactions-12] killing on exit
[master-11] killing on exit
[turtlebot_laptop_battery-8] killing on exit
[cmd_vel_mux-7] killing on exit
[bumper2pointcloud-6] killing on exit
[mobile_base-5] killing on exit
[capability_server-9] killing on exit
[mobile_base_nodelet_manager-4] killing on exit
[diagnostic_aggregator-3] killing on exit
[robot_state_publisher-2] killing on exit
Exception in thread Thread-25 (most likely raised during interpreter shutdown):
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
  File "/usr/lib/python2.7/threading.py", line 763, in run
  File "/opt/ros/indigo/lib/python2.7/dist-packages/bondpy/bondpy.py", line 247, in _publishing_thread
  File "/usr/lib/python2.7/threading.py", line 215, in __exit__
  File "/usr/lib/python2.7/threading.py", line 202, in release
<type 'exceptions.TypeError'>: 'NoneType' object is not callable
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

gaschler, 2017-12-08:
First, turtlebot is part of the cartographer_turtlebot project, so `roslaunch cartographer_turtlebot turtlebot.launch` (not _ros).

trajectory_builder.lua is a configuration file provided by cartographer.
Also make sure you follow the catkin install instructions, and do not forget `source install_isolated/setup.bash`.

I'm believe this is resolved, feel free to re-open if necessary.

closed 2017-12-08 by gaschler

araster1010, 2018-01-16:
I have a same problem.

And if I run  `roslaunch cartographer_turtlebot turtlebot.launch` first, the console says below.

[/home/(user name)/catkin_ws/install_isolated/share/cartographer_turtlebot/launch/turtlebot.launch] requires the 'configuration_basename' arg to be set
The traceback for the exception was written to the log file

I'm a beginner at ROS. Could you tell me the meaning of 'configuration_basename' arg  and what shall I do?

Thanks.

gaschler, 2018-01-16:
configuration_basename is "Basename, i.e. not containing any directory prefix, of the configuration file."
For example, backpack_2d.lua.
Follow https://google-cartographer.readthedocs.io/en/latest/index.html, start with the launch files included and go from there.