cartographer-project/cartographer_turtlebot #83: demo_lidar_2d.launch cartographer_turtlebot_demo.bag shows double walls

gaschler, 2018-01-03:


gaschler, 2018-01-22:
@SirVer Do you have an idea if this got worse (is a regression)? Or am I using the wrong launch file for this bag?

SirVer, 2018-01-23:
I think this regressed a bit. 3D depth camera data has always been finicky though.

SubMishMar, 2018-06-25:
@SirVer, In case I do not want to use odometry and imu data, how should I set 
`  map_frame = "map",
  tracking_frame = "camera_rgb_frame",
  published_frame = "odom",
  odom_frame = "odom",`
In order to get consistent results. Right now, I have set the flags for using odometry and imu to false and with the above frame values my result looks completely wrong. From what it seems I need to fix these frames properly but I dont exactly understand how, even after reading your documentation. Can you please throw some light on this?