cartographer-project/cartographer_turtlebot #89: the time of imu error

gavinwangpj, 2018-05-02:
hello everyone, i have run cartographer in turtlebot3 waffle pi, but can't use imu to make a map. no_imu is ok. and the error log as follow:
[ INFO] [1525244437.574660587]: I0502 15:00:37.000000  3937 collated_trajectory_builder.cc:72] scan rate: 4.92 Hz 2.03e-01 s +/- 2.32e-02 s (pulsed at 99.70% real time)
t_:636608412387287190, t: 636608412387222648,dt=-0.0064542
F0502 15:00:38.786036  3937 ordered_multi_queue.cc:112] Check failed: last_dispatched_time_ <= next_data->GetTime() (636608412387287190 vs. 636608412387222648) Non-sorted data added to queue: '(0, imu)'
[FATAL] [1525244438.787813602]: F0502 15:00:38.000000  3937 ordered_multi_queue.cc:112] Check failed: last_dispatched_time_ <= next_data->GetTime() (636608412387287190 vs. 636608412387222648) Non-sorted data added to queue: '(0, imu)'
*** Check failure stack trace: ***
    @ 0x76eb986e  google::LogMessage::Fail()
    @ 0x76ebae6a  google::LogMessage::SendToLog()
    @ 0x76eb951c  google::LogMessage::Flush()
    @ 0x76ebb48c  google::LogMessageFatal::~LogMessageFatal()
    @   0x1bacd8  cartographer::sensor::OrderedMultiQueue::Dispatch()
    @   0x1bb482  cartographer::sensor::OrderedMultiQueue::Add()
    @   0x1bc3e0  cartographer::sensor::Collator::AddSensorData()
    @   0x21280e  cartographer::mapping::CollatedTrajectoryBuilder::AddData()
    @   0x213c8c  cartographer::mapping::CollatedTrajectoryBuilder::AddSensorData()
    @   0x1a1bd8  cartographer_ros::SensorBridge::HandleOdometryMessage()
    @   0x183fc0  cartographer_ros::Node::HandleOdometryMessage()
    @   0x18d468  boost::detail::function::void_function_obj_invoker1<>::invoke()
    @   0x195520  ros::SubscriptionCallbackHelperT<>::call()
    @ 0x76772bbc  ros::SubscriptionQueue::call()
    @ 0x767382e4  ros::CallbackQueue::callOneCB()
    @ 0x76738fa0  ros::CallbackQueue::callAvailable()
    @ 0x76775a24  ros::SingleThreadedSpinner::spin()
    @ 0x767650fc  ros::spin()
    @   0x17ecf6  cartographer_ros::(anonymous namespace)::Run()
    @   0x17c54e  main
    @ 0x763668aa  __libc_start_main
[cartographer_node-1] process has died [pid 3937, exit code -6, cmd /home/sunny/gcatkin_ws/install_isolated/lib/cartographer_ros/cartographer_node -configuration_directory /home/sunny/catkin_ws/src/turbot3/turbot3_slam/config -configuration_basename turbot3_lidar_2d.lua scan:=/scan __name:=cartographer_node __log:=/home/sunny/.ros/log/dea8386a-4dcb-11e8-9d35-e470b8ea7c52/cartographer_node-1.log].
log file: /home/sunny/.ros/log/dea8386a-4dcb-11e8-9d35-e470b8ea7c52/cartographer_node-1*.log

sradmard, 2018-07-04:
You may want to check this issue:
[https://github.com/googlecartographer/cartographer_ros/issues/699](url)

justin-changqi, 2021-04-14:
Hi @sradmard, how do you print the FATAL messages? ([FATAL] [1525244438.787813602]: F0502 15:00:38.000000 3937 ordered_multi_queue.cc:112] ... in your case). I fail to run the demo_backpack_2d.launch with the official bag file on Ubuntu20.04, and it showed "[cartographer_node-2] process has died [pid 156314, exit code -11, cmd....". It is hard to find out what happed if it does not show the FATAL message.

Milalaa, 2022-03-29:
hi！@sradmard  I meet the same problem. How do you solve it ?