cartographer-project/cartographer_turtlebot #90: demo_depth_camera_2d.launch and turtlebot_depth_camera_2d.launch fail to launch

nfopma, 2018-06-08:
Thanks for the great package! It looks very promising and I can not wait to get it to work on my Turtlebot. 

Unfortunately, for the last 2 days I have been struggling with the actual deployment. I followed the instructions of [installing cartographer_ros ](https://google-cartographer-ros.readthedocs.io/en/latest/)and [installing cartographer_turtlebot](http://google-cartographer-ros-for-turtlebots.readthedocs.io/en/latest/). I am able to run all of the demo's, except for demo_depth_camera_2d.launch. I am using Ubuntu 16.04 and ROS kinetic. 

I seem to be getting the exact same errors when I try to launch turtlebot_depth_camera_2d.launch. Launching turtlebot_depth_camera_3d.launch works perfectly without errors, and I can see the 3D map being built in Rviz when I drive around the Turtlebot. Therefore, the problem seems to be specifically about the 2D mapping.  

I put the terminal output I get in two txt files. 

[cartographer_terminal_output_2d_turtlebot.txt](https://github.com/googlecartographer/cartographer_turtlebot/files/2084451/cartographer_terminal_output_2d_turtlebot.txt)
[cartographer_terminal_output_2d_demo.txt](https://github.com/googlecartographer/cartographer_turtlebot/files/2084446/cartographer_terminal_output_2d_demo.txt)

For both files, the following is highlighted in red in the terminal: 

`[FATAL] [1528462899.578349680]: F0608 15:01:39.000000 16788 lua_parameter_dictionary.cc:410] Check failed: 1 == reference_counts_.count(key) (1 vs. 0) Key 'resolution' was used the wrong number of times.`

And: 

`[cartographer_node-2] process has died [pid 16788, exit code -6, cmd /home/nathan/ros_ws/install_isolated/lib/cartographer_ros/cartographer_node -configuration_directory /home/nathan/ros_ws/install_isolated/share/cartographer_turtlebot/configuration_files -configuration_basename turtlebot_depth_camera_2d.lua __name:=cartographer_node __log:=/home/nathan/.ros/log/14f00552-6b1c-11e8-b200-a45e60d3ab15/cartographer_node-2.log].
log file: /home/nathan/.ros/log/14f00552-6b1c-11e8-b200-a45e60d3ab15/cartographer_node-2*.log`

The only difference between the Turtlebot 2D launchfile vs. 3D launchfile  seem to be the the passing of [turtlebot_depth_camera_2d.lua](https://github.com/googlecartographer/cartographer_turtlebot/blob/master/cartographer_turtlebot/configuration_files/turtlebot_depth_camera_2d.lua) vs. the passing of [turtlebot_depth_camera_3d.lua
](https://github.com/googlecartographer/cartographer_turtlebot/blob/master/cartographer_turtlebot/configuration_files/turtlebot_depth_camera_3d.lua). 

If anyone could help me with understanding and work towards solving this error I would greatly appreciate it!

nfopma, 2018-06-08:
@gavanderhoorn

SubMishMar, 2018-06-23:
Comment out
https://github.com/googlecartographer/cartographer_turtlebot/blob/f26a7246f0e2774e8f9658d8178a7a77420c22e6/cartographer_turtlebot/configuration_files/turtlebot_depth_camera_2d.lua#L57
 and 
https://github.com/googlecartographer/cartographer_turtlebot/blob/f26a7246f0e2774e8f9658d8178a7a77420c22e6/cartographer_turtlebot/configuration_files/turtlebot_depth_camera_3d.lua#L59-L60

Then build and install the code by typing `catkin_make_isolated --install --use-ninja` in the ros workspace, then source the workspace. run the launch file again, it should work!

gavanderhoorn, 2018-06-23:
Are the 3D parameters used in the 2d demo?

SubMishMar, 2018-06-25:
@gavanderhoorn I don't think so. But commenting out those params helped me run the 3D demo as well.

nfopma, 2018-06-26:
Thanks for the reply @SubMishMar! 

I tried to remove the resolution parameter from the configuration file, but when I run the depth camera 2D demo with the provided bag I get the following mapping result: 

[image]

Compared to the presumed correct mapping when running the LIDAR demo: 

[image]

Did you have different results?

SubMishMar, 2018-06-26:
@nfopma I get the same results as you do. The result with 2D depth is not great and with LIDAR it gets a lot better.

gavanderhoorn, 2018-06-26:
@SubMishMar wrote:
> The result with 2D depth is not great [..]

that is a bit of an understatement :)

gaschler, 2018-07-04:
Fixed by #95

closed 2018-07-04 by gaschler

nfopma, 2018-07-11:
Unfortunately, the [proposed changes](https://github.com/googlecartographer/cartographer_turtlebot/pull/95/commits/37645fc4536ded798ade613af2237d1a70c1a8f0) did not significantly improve the results that I got when I ran the 2d depth camera demo. 

[image]