cartographer-project/cartographer_turtlebot #93: The existence of key 'TRAJECTORY_BUILDER_2D.submaps.resolution'

tongtybj, 2018-07-02:
Also reported by #90 , I can not launch the turtlebot  with ``` turtlebot_depth_camera_2d.launch```, unless I comment out [ ```TRAJECTORY_BUILDER_2D.submaps.resolution```](https://github.com/googlecartographer/cartographer_turtlebot/blob/master/cartographer_turtlebot/configuration_files/turtlebot_depth_camera_2d.lua#L57).

The following error log indicates that this parameter is not included in the cartographer dictionary. 
Can anyone tell me why this unused parameter is still left in the config file?
```
[FATAL] [1530551123.368644980]: F0703 02:05:23.000000 29119 lua_parameter_dictionary.cc:410] Check failed: 1 == reference_counts_.count(key) (1 vs. 0) Key 'resolution' was used the wrong number of times.
```

gaschler, 2018-07-11:
Closing as fixed by #95

closed 2018-07-11 by gaschler

tongtybj, 2018-07-11:
@gaschler Thanks!