cartographer-project/cartographer_turtlebot #99: cartographer installation error

z3y50n, 2019-03-15:
I am doing all the steps pointed at [https://google-cartographer-ros-for-turtlebots.readthedocs.io/en/latest/#](https://google-cartographer-ros-for-turtlebots.readthedocs.io/en/latest/#).
During halfway of installation it gives this error:

> [118/121] Building CXX object absl/synchronization/CMakeFiles/absl_synchronization.dir/mutex.cc.o
[119/121] Linking CXX static library absl/synchronization/libabsl_synchronization.a
[120/121] Building CXX object absl/debugging/CMakeFiles/absl_failure_signal_handler.dir/failure_signal_handler.cc.o
[121/121] Linking CXX static library absl/debugging/libabsl_failure_signal_handler.a
[18/391] Running C++ protocol buffer c...form/proto/timestamped_transform.proto
FAILED: cd /home/r4a/catkin_cartographer/build_isolated/cartographer/install && /usr/bin/protoc --cpp_out /home/r4a/catkin_cartographer/build_isolated/cartographer/install -I /home/r4a/catkin_cartographer/src/cartographer /home/r4a/catkin_cartographer/src/cartographer/cartographer/transform/proto/timestamped_transform.proto
cartographer/transform/proto/timestamped_transform.proto:15:10: Unrecognized syntax identifier "proto3".  This parser only recognizes "proto2".
[18/391] Running C++ protocol buffer c...her/mapping/proto/3d/hybrid_grid.proto
FAILED: cd /home/r4a/catkin_cartographer/build_isolated/cartographer/install && /usr/bin/protoc --cpp_out /home/r4a/catkin_cartographer/build_isolated/cartographer/install -I /home/r4a/catkin_cartographer/src/cartographer /home/r4a/catkin_cartographer/src/cartographer/cartographer/mapping/proto/3d/hybrid_grid.proto
cartographer/mapping/proto/3d/hybrid_grid.proto:15:10: Unrecognized syntax identifier "proto3".  This parser only recognizes "proto2".
[18/391] Running C++ protocol buffer c...graph/constraint_builder_options.proto
FAILED: cd /home/r4a/catkin_cartographer/build_isolated/cartographer/install && /usr/bin/protoc --cpp_out /home/r4a/catkin_cartographer/build_isolated/cartographer/install -I /home/r4a/catkin_cartographer/src/cartographer /home/r4a/catkin_cartographer/src/cartographer/cartographer/mapping/proto/pose_graph/constraint_builder_options.proto
cartographer/mapping/proto/pose_graph/constraint_builder_options.proto:15:10: Unrecognized syntax identifier "proto3".  This parser only recognizes "proto2".
[18/391] Running C++ protocol buffer c...grapher/mapping/proto/pose_graph.proto
FAILED: cd /home/r4a/catkin_cartographer/build_isolated/cartographer/install && /usr/bin/protoc --cpp_out /home/r4a/catkin_cartographer/build_isolated/cartographer/install -I /home/r4a/catkin_cartographer/src/cartographer /home/r4a/catkin_cartographer/src/cartographer/cartographer/mapping/proto/pose_graph.proto
cartographer/mapping/proto/pose_graph.proto:15:10: Unrecognized syntax identifier "proto3".  This parser only recognizes "proto2".
ninja: build stopped: subcommand failed.
<== Failed to process package 'cartographer': 
  Command '['/home/r4a/catkin_cartographer/install_isolated/env.sh', 'ninja', '-j4', '-l4']' returned non-zero exit status 1

We are using turtlebot 2

joanvallve, 2019-11-25:
The same happens to me when I try to install it in a PC using Ubuntu 16 & ROS Kinetic.

joanvallve, 2019-11-25:
Seems that installing v3 of protobuf for c++ solves this issue: https://github.com/protocolbuffers/protobuf/blob/master/src/README.md

This should be handled by ROSDEP or at least should be included in the installation procedure.

MichaelGrupp, 2020-08-28:
Yes, you need protobuf 3. This shouldn't be an issue with newer Ubuntu distros anymore.

closed 2020-08-28 by MichaelGrupp