cartographer-project/cartographer_turtlebot #102: change topic name from "laser_scan" to "scan"

dwindy, 2020-02-26:
Hi
I successively run the cartographer_trutlebot_demo.bag.
Noticed in that bag, the scan information was published under the topic "/laser_scan".
I recorded my own bag, trying to run the cartographer_turtlebot based on that.
The terminal outputs
[ WARN] [1582745849.330702955]: W0226 19:37:29.000000  2762 node.cc:857] Expected topic "scan" (trajectory 0) (resolved topic "/laser_scan") but no publisher is currently active.
[ WARN] [1582745849.330925222]: W0226 19:37:29.000000  2762 node.cc:857] Expected topic "odom" (trajectory 0) (resolved topic "/odom") but no publisher is currently active.

In my bag, the scan topic name is "/scan" is there any way to fix that?
what's more, I have "/Odom" topic in my bag but the terminal still said no publisher is currently active.

Thank you

dwindy, 2020-02-26:
Well, I

closed 2020-02-26 by dwindy

liushuai0901, 2024-05-22:
我和你一样遇到了相同的问题，请问您解决了吗？如果解决了可以教教我怎么解决这个问题吗？